f32c/f32c #11: Disabling the multiplier and the barrel shifte

goran-mahovlic, 2015-07-15:
Kazam from numato forum

Hi,

I have some more time now and want to get this working. :)

Looking over the VHDL I have two questions:

1. I tried disabling the multiplier and the barrel shifter to see how much area could be saved. This reduced the utilization from 57% to 50% of slices. Is this recommended or discouraged?

2. As I understand it the BRAM is generated from slices in the FPGA but can I use the BRAM generator in ISE to instead use the dedicated BRAM resources to reduce utilization?

https://numato.com/community/threads/arduino-on-numato-fpga-boards.26/page-3#post-229

gornjas, 2015-07-15:
Disabling the multiplier requires compiling and linking with a library with software multiplication (MULT instructions disabled). Such a library is already available in f32c/src, but we haven't integrated it in FPGArduino (yet). This applies only to MI32 binaries, since RV32I binaries never include multiplication instructions, so for C_arch = ARCH_RV32 setting C_mult_enable to false has no effect.

Barrel shifter may be (partially) disabled to save a few LUTs on both MI32 and RV32 without recompiling or relinking, the only tradeoff is slower (iterative) shifting by 1 to 7 places. Shifts by 8, 16 and 24 places are still performed in a single cycle.

As of the BRAM generator in ISE, I had nothing but trouble with it, so you are free to try it out yourself, but be warned not to expect much help from us on that endeavour. Custom 6K BRAM modules for s3A50 containing MI32 bootloader just hit the github repo, so I'd encourage you to use that.

Note that RV32 configuration just barely fits into S3A50, though we couldn't get the serial communication to work via the onboard USB PIC. The MI32 with should also fit as of today (consuming 702 of 704 available slices), though again that variant uses a reduced MI32 instruction subset which calls for a specialized library, which we hope to prepare shortly.

goran-mahovlic, 2015-07-15:
tnx

closed 2015-07-15 by goran-mahovlic