f32c/f32c #12: s3_toplevel_bram

goran-mahovlic, 2015-07-15:
if I want to use s3_toplevel_bram for s3_250 I got error   
 Number of RAMB16s:                     17 out of      12  141% (OVERMAPPED)
I know if I change C_mem_integer it will pass compailing
	C_mem_size: integer := 16;
Is the better way of solving this to change it in s3_toplevel_bram so all will have 16 or like now to add new top level or you can open some simple file where we could define this?

gornjas, 2015-07-15:
The s3_toplevel_bram is intended to be used for Xilinx / Digilent Spartan 3E 500 / 1600 and 3A / 3AN 700 / 1400 boards, which all have very similar layouts and on-board peripherals (LCD module, rotary encoder, buttons, switches, two serial ports etc.). All the mentioned boards have sufficient BRAM resources to be configured with C_mem_size = 32. So I'd recommend using a separate top-level module for the s3e-250 board, the one already in the tree but it should be trimmed of unnecessary I/O signals (which obviously originate from C/P from s3_toplevel_bram.vhd).

goran-mahovlic, 2015-07-15:
tnx

closed 2015-07-15 by goran-mahovlic