f32c/f32c #13: DDR SDRAM support

goran-mahovlic, 2015-07-16:
http://numato.com/tutorials/learning-fpga-and-verilog-a-beginners-guide-part-6-ddrsdram/
http://opencores.org/project,ddr_sdr

emard, 2016-04-07:
ESA11 compiles DDR3 support with axi_interconnect
acram.vhd is used to interface with ESA11 axi_cache.vhd
RAM interface

there's also acram_emu.vhd which emulates axi_cache
ram interface using BRAM modules

emulation works on spartan6 at 100MHz as a proof of concept,
but it does not work on 7a35i (searching for reasons)
 
acram.vhd doesn't work on 7a35i over axi_cache

We are close but still no success

emard, 2016-05-01:
Thanks to ESA11 team who sent hardware and some sources,
DDR3 RAM support started to work!

Currently it works with mig-7 and axi-4 interconnect on vivado
2016.1 compiled artix-7 xc7a35i for ESA11

closed 2016-05-01 by emard