f32c/f32c #18: MimasV2 project won't build

Ka-zam, 2015-08-07:
ISE 14.7 error output is:

ERROR:HDLCompiler:539 - "/home/magnus/src/f32c/rtl/soc/fm/fmgen.vhd" Line 95: Indexed name is not a std_logic_vector
ERROR:HDLCompiler:854 - "/home/magnus/src/f32c/rtl/soc/fm/fmgen.vhd" Line 44: Unit <x> ignored due to previous errors.

goran-mahovlic, 2015-08-07:
I can try it when I get back home it will be at monday
I did succesfully compile it with 115200,  FM VGA ... if you want you can try bin

https://github.com/goran-mahovlic/f32c_extras/blob/master/Binaries/MimasV2/MimasV2_FPGArduino_115200_initial_SPI_tested_working_VGA_RDS.bin

Ka-zam, 2015-08-07:
Hi, tried your binary but I couldn't get it to work. All I get is a blue LED that constantly stays on.

emard, 2015-08-13:
I fixed issue in glue_bram_vga.vhd regarding propagation of FM DDS clock
frequency, made fm.vhd (handles other modules from soc/fm/*)
as a single entity to be instantiated from glue.
It workst on scarab bram hdmi see which is spartan-6 similar to numato mimas
some copy/paste from there should get mimas compiled and running

goran-mahovlic, 2015-08-13:
can you please try to build it now, I have some license issue and build fails but it may work on other computer...

goran-mahovlic, 2015-08-13:
I have solved license issue and now it compiles ok I will test it after 17:00
https://github.com/goran-mahovlic/f32c_extras/blob/master/Binaries/MimasV2/MimasV2_FPGArduino_115200_initial_SPI_VGA_RDS_HDMI.bin

goran-mahovlic, 2015-08-13:
For now I can confirm that this bin is working, I have fade leds, VGA test, black screen on HDMI did not test serial yet

goran-mahovlic, 2015-08-13:
It now compiles ok, and I can upload from arduino at 115200

closed 2015-08-13 by goran-mahovlic

Ka-zam, 2015-08-16:
I can confirm this is working on my board and I have the blinking LED.