f32c/f32c #39: lattice FX2 (ULX2S) textmode not fully functional

emard, 2015-12-21:
In the textmode soft scroll example, corruption occurs
in the font 8-bit BRAM

pixels slowly change without reason 
(even with bram write signal disabled)

textmode bitmap doesn't work at all.
(thus for ulx2s disabled)

vgahdmi bitmap works.

emard, 2015-12-24:
Bitmap "solution"

There was degenerate if comparison
std_logic_vector(-1 downto 0)  = std_logic_vector(-1 downto 0)
on xilinx holds true but on lattice is false
one OR fixed it.

It shows bitmap but textmode still won't
go faster than 50 MHz and there's no
enough memory bandwidth for 8bpp bitmap
so betmap although now is shown, is split
apart due to low bandwidth

XarkLabs, 2015-12-24:
Hmm, XO2 has trouble > 50Mhz also so not all that surprising perhaps.

emard, 2015-12-24:
One thing is strange, I use vgahdmi with bitmap only
and change clock
81.25MHz works
75MHz doesn't work (can't upload)
62MHz doesn't work (can't upload)
50MHz works but screen garbled (low bandwidth)

emard, 2016-08-21:
Due to complexity of vga textmode, 50MHz is fmax on lattice FX2

closed 2016-08-21 by emard