f32c/f32c #46: Arduino IED eror: ftdi_set_latency_timer() failed

asukiaaa, 2016-04-30:
The following error occurred when I tried to write a sample program of Arduino to my miniSpartan6+ from ubuntu14.04.
```
TX error at 80000000
ftdi_set_latency_timer() failed
```
[image]

I tried USB0 and USB1 but both occurred the error.

However, it was succeeded that a direct writing with the following command.
```
$ sudo xc3sprog -c ftdi ~/.arduino15/packages/FPGArduino/tools/bootloaders/1.0.0/scarab/scarab_sdram_xc6slx25_riscv_vgabit_100mhz.bit
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 768 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop

Using Libftdi, 
DNA is 0x198231f2fa06f9fe
```

Is there any idea to solve the error?
Thank you.

emard, 2016-05-01:
Ttry to recompile ujprog from source, try different computer or
debian instead of ubuntu

emard, 2016-05-02:
HI

Something is wrong with scarab bitstreams, I should check them
please standby

emard, 2016-05-02:
HI

In the boards manager I refreshed the mainline set of bistreams
for scarab SDRAM MIPS 100 MHz little endian set of bitstreams.

Now I can upload (before it wasn't possible)

Try it and let me know how it works now

asukiaaa, 2016-05-02:
> Ttry to recompile ujprog from source
> Also try HEX protocol

Thanks for your advice.
Uploading with compiled ujprog and HEX protocol mode finished without any error.
I did make and installed ujprog with the following commands.
```
git clone git@github.com:f32c/tools.git
cd tools/ujprog
cp Makefile.linux makefile
make
sudo make install // maybe no meaning in this case
cp ujprog ~/.arduino15/packages/FPGArduino/tools/ujprog/1.0.0/ujprog
```

However, nothing happens to leds on my miniSpartan6+.
(The leds changed if I upload bit file by command)

To check compile of ujprog was succeeded or not, is there any way to test it?

> Something is wrong with scarab bitstreams, I should check them
> In the boards manager I refreshed the mainline set of bistreams

[image]

Thank you but leds on a board still not change.
I updated Arduino IED by following flow.
Is the flow is enough to update?

1. Removed FPGArduino from Board Manager of Arduino IDE.
2. Removed files in ~/.arduino15/staging/packages/
3. Install FPGArduino with using Board Manager of Arduino IED.

Thanks.

emard, 2016-05-02:
HI

Your update flow is correct.
Your compilation of ujprog also passed correctly.

After selecting Scarab board and "burn bootloader" you should
see fading leds. Afater that bootloader burn, /dev/ttyUSB0 will disappear
and remain /dev/ttyUSB1 which you could select by port

One simple problem might also be unix permission
try as root

chmod a+rw /dev/ttyUSB1

asukiaaa, 2016-05-02:
Thank you, I got it!
I saw fading led after "burn bootloader" and saw led blink after update "Blink" sketch.

Please allow me to ask a additional question about permission and leave memo to pass problems for other ubuntu users.

**Question.**
To use FPGArduino, do I always need to run arduino under sudo mode?

**Memo**
First problem.
```
/home/asuki/.arduino15/packages/FPGArduino/tools/openocd/0.9.0/bin/openocd: error while loading shared libraries: libhidapi-hidraw.so.0: cannot open shared object file: No such file or directory
Error while burning bootloader.
```
 I passed it by installing a library.

```
sudo apt-get install libhidapi-dev
```

Secound problem.
```
Open On-Chip Debugger 0.9.0 (2015-06-17-22:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop

Could not open FTDI device (using libftdi): inappropriate permissions on device!
Unable to access FTDI device with either libftdi or FTD2XX
Error while burning bootloader.
```

I succeeded with running arduino by sudo.
I did not change chmod setting.

```
sudo ln -s /home/asuki/.arduino15 /root/
sudo ~/arduino-1.6.8/arduino
```

emard, 2016-05-02:
HI

Great that you got blink led running! Plug HDMI monitor and
test some examples from f32c_vgatextconsole too :) 

In the https://github.com/f32c/arduino I have attached
udev rule that will set permissions automatically when
you plug-in usb connector.

You will no longer need to run  arduino, ujprog, xc3sprog or openocd as root.

closed 2016-05-02 by emard

asukiaaa, 2016-05-03:
> Plug HDMI monitor and test some examples from f32c_vgatextconsole too :)
Wao, it's interesting!

> You will no longer need to run arduino, ujprog, xc3sprog or openocd as root.
I expect for your job.

(I'm happy if I can start FPGArduino without writing some sketch of Arduino IDE to FPGA.
 Now I can start from bootloader fading.)

Thanks a lot!

roman3017, 2016-08-07:
After following this case with the same configuration and arduino version I am still not able to burn bootloader. Here is the error:

`
Open On-Chip Debugger 0.9.0 (2015-06-17-22:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop
Using Libftdi, 
DNA is 0xf98ebbedb4639dfe
Device failed to configure, INSTRUCTION_CAPTURE is 0x19
`

I am getting INSTRUCTION_CAPTURE is 0x19 even when I use xc3sprog from the command line:

`
xc3sprog -c ftdi ~/.arduino15/packages/FPGArduino/tools/bootloaders/1.0.0/scarab/scarab_sdram_xc6slx25_riscv_vgabit_100mhz.bit 
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop
Using Libftdi, 
DNA is 0xf98ebbedb4639dfe
Device failed to configure, INSTRUCTION_CAPTURE is 0x19
`

Do you know how to fix this error?

emard, 2016-08-07:
HI

this is maybe nothing, just unix device permission access rights
you might try as root of course but also ordinary user member
of "dialout" group

create this file in this directory, reboot linux and try again

# /etc/udev/rules.d/80-fpga-scarab.rules
# this is for usb-serial tty device
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", \
  MODE="664", GROUP="dialout"
# this is for ujprog libusb access
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", \
  GROUP="dialout", MODE="666"

roman3017, 2016-08-07:
Thanks for prompt response. I have that file and I am member of dialout group.

roman3017, 2016-08-07:
`
sudo xc3sprog -c ftdi ~/.arduino15/packages/FPGArduino/tools/bootloaders/1.0.0/scarab/scarab_sdram_xc6slx25_riscv_vgabit_100mhz.bit 
[sudo] password for roman: 
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop
Using Libftdi, 
DNA is 0xf98ebbedb4639dfe
Device failed to configure, INSTRUCTION_CAPTURE is 0x19
`

emard, 2016-08-07:
You might have a different release of scarab board than I have.
Either different xc6slx25 or ft2232 chip

this option will dump xc3sprog internal devices it could recognize
xc3sprog -D
into 2 files, one covering xilinx chips, other covering cable drivers
(onboard ft2232 belongs to cables)

can you try to check is your chips are actually listed there
you can edit this files and starting xc3sprog at the same directory where
the files are, maybe experimenting a bit

this will check connection to ft2232

xc3sprog -c ftdi

should print something similar to this

XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop

Using Libftdi,
JTAG loc.:   0  IDCODE: 0x13631093  Desc:
XA7A100T Rev: A  IR length:  6

If nothing works one possibility is to download or recompile latest xc3sprog

roman3017, 2016-08-07:
I have recompiled latest xc3prog version 785:

`
$ ./xc3sprog -D
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 785 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop
Dumping internal cablelist to cablelist.txt
Dumping internal devicelist to devlist.txt
`
`
$ ./xc3sprog -c ftdi XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 785 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop
Using Libftdi, 
JTAG loc.:   0  IDCODE: 0x24001093  Desc:                        XC6SLX9 Rev: C  IR length:  6
`

roman3017, 2016-08-07:
Oh, I see, I bet it is because of this:  XC6SLX9.

roman3017, 2016-08-07:
Is it possible to support XC6SLX9 version of minispartan6+?

emard, 2016-08-07:
HI

OK your board has xc6slx9 :)

Yes support for this chip is very possible and should be no problem
I guess just changing chip name in the makefile and recompiling f32c,
slx9 works on many other boards

send me exactly what is written on your chip, all the lines printed
on the chip also thin writing of the speed grade too if you can read it

I can try to recompile and send you test bitstream tomorrow

roman3017, 2016-08-07:
Thank you very much for your help:

XC6SLX9
FTG256BIV1421
D5003032A
3C

roman3017, 2016-08-07:
Are there any instructions on how to recompile f32c? Is it just making this file?
f32c/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi_generic/Makefile

roman3017, 2016-08-07:
I have got it working now:

`
$ cd f32c/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/
`

`
$ git diff
`
`
diff --git a/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile b/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile
`
`
index b053c92..3580cc6 100644
`
`
--- a/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile
`
`
+++ b/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile
`
`
@@ -1,7 +1,8 @@
`
`
+MODEL ?= xc6slx25
`
`
 project = miniSpartan6-plus
`
`
 vendor = xilinx
`
`
 family = spartan6
`
`
-part = xc6slx25-ftg256-3 # last number -3 is speed grade
`
`
+part = ${MODEL}-ftg256-3 # last number -3 is speed grade
`
`
 flashsize = 8192 # KB (kilobytes) - SPI flash device is M25P40
`
`
 top_module = scarab_xram_sdram
`
`
 isedir = /opt/Xilinx/ISE/14.7/ISE_DS
`

`
$ MODEL=xc6slx9 make clean
`
`
$ MODEL=xc6slx9 make
`
`
$ xc3sprog -c ftdi miniSpartan6-plus.bit
`

Afterwards I could successfully upload and run blinky via /dev/ttyUSB1. Thank you very much for your help.

emard, 2016-08-07:
Hey, That are great news!

I'll include you patch to the Makefile
If you have chance, try to connect montor HDMI, how does
picture look?. Both HDMI ports should output picture, they
share the same output drivers.

on my board and monitor, port labeled as "output" has less
flickers then the port labeled as "input".

roman3017, 2016-08-07:
I can see 5 horizontal stripes (white, green, white, purple) on both HDMI ports after uploading miniSpartan6-plus.bit. They look the same. 

Unfortunately, I cannot compile framebuffer exsample in IDE, since I got this error:

.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/libraries/f32c/examples/framebuffer/framebuffer.ino:9:16: fatal error: fb.h: No such file or directory
 #include \<fb.h\>
                ^
compilation terminated.
exit status 1
Error compiling for board Scarab miniSpartan6+ (Xilinx Spartan-6 XC6SLX25, 64K BRAM, 81.25/112.5 MHz).

emard, 2016-08-07:
One note:

recent git you have just compiled has changed semantics
for gpio and timer interrupt to be similar to AVR and some
bugfix also. So some examples mostly under (f32c/interrupts)
wont work with old arduino library

for this, arduino needs to recently pull from the git, f32c/arduino
and manually copy/replace the directories "core" and "libraries"
under ~/.arduino15

roman3017, 2016-08-07:
I can compile arcagol and see it on both ports. Great, both ports look same.

emard, 2016-08-07:
Stripes are great, as long as they are stable and
don't flicker, you are looking at unintialized RAM

framebuffer is obsolete and only for old analog TV inputs

try stuff at examples/Compositing  and examples/f32c_VGATextMode
there's new video action with tiled graphics and sprite 2D acceleration

emard, 2016-08-07:
As both ports look the same I think your compiled bitstream is
completely functional for some experimentation

scarab build is pretty full-featured, f32c on lx9 is same as lx25 for
everything except too small for full featured FPU vector processor

roman3017, 2016-08-07:
I can run examples from f32c_VGATextMode. Some are better than others. Really good one is f32c_compositing. Both my ports behave the same as far as I can tell. Uninitialized RAM is very stable with no flickering of the stripes although stripes have few characters there. I am really glad it is working but only in MIPS mode. When I try to compile anything as RISCV I get warning:

.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/WInterrupts.c:305:2: warning: #warning "RISC-V doesn't have interrupts yet. Choose Tools->CPU Architecture->MIPS" [-Wcpp]
 #warning "RISC-V doesn't have interrupts yet. Choose Tools->CPU Architecture->MIPS"
  ^
Sketch uses 968 bytes (0%) of program storage space. Maximum is 33,521,664 bytes.

Sketch does not work when uploaded.

emard, 2016-08-07:
HI

RISC-V is partially broken/under_construction_by_Marko,

it won't work with external RAM.

RISC-V can only work if it is compiled as a project which uses
BRAM only and no interrupt support yet also.

MIPS ISA performs faster than RISC-V ISA on same hardware.

roman3017, 2016-08-07:
Thanks for your explanation. I have installed FPGArduino just today. Does json script pull latest changes?

ls ~/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/
boards.txt  **libraries**     programmers.txt  tools
**cores**       platform.txt  system           variants

ls
hardware  **libraries**  README.md

ls hardware/fpga/f32c/
boards.txt  **cores**  platform.txt  programmers.txt  system  tools  variants

Should I copy and replace highlighted folders?

emard, 2016-08-07:
I mean you need to compile bitstream for RISC-V
then uploading the sketch might work (I haven't tried that
for a long time :)
so in the top module
xilinx/scarab_ms6p/top/top_xram_sdram.vhd
change this lines
C_arch: integer := ARCH_MI32;
C_sdram: boolean := true;
into
C_arch: integer := ARCH_RV32;
C_sdram: boolean := false;

I don't have proper RISC-V project, maybe other things need
to adjusted also

and make clean; MODEL=xc6slx9 make
the new bitstream for RISC-V

roman3017, 2016-08-07:
I see, I will try and let you know. Thanks.

emard, 2016-08-07:
reply to prev message, yes hightlighted folders need to be replaced
for MIPS interrupts to work

but still we have integer multiply bug in interrupt context
so each ISR must avoid to execute integer multiply...

emard, 2016-08-07:
so json doesn't pull latest git changes
I'm delaying binary json release because I need to recompile many bistreams
and axi vector dma is still not working

so fix is to manually replace directory found on git's f32c/arduino
into your json auto-installed stuff, you will recognize that's correct
directory becaouse they have similar file names inside

~/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores
~/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/libraries

roman3017, 2016-08-07:
Ok, I have replaced arduino folders and recompiled bitstream. Unfortunately, after uploading bitstream the screen is blank on both ports and even simplest blinkey sketch does not work. I guess riscv has to wait...

$ git diff
diff --git a/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile b/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile
index b053c92..3580cc6 100644
--- a/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile
+++ b/rtl/proj/xilinx/scarab/scarab_ms6p_xram_sdram_hdmi/Makefile
@@ -1,7 +1,8 @@
+MODEL ?= xc6slx25
 project = miniSpartan6-plus
 vendor = xilinx
 family = spartan6
\-part = xc6slx25-ftg256-3 # last number -3 is speed grade
\+part = ${MODEL}-ftg256-3 # last number -3 is speed grade
 flashsize = 8192 # KB (kilobytes) - SPI flash device is M25P40
 top_module = scarab_xram_sdram
 isedir = /opt/Xilinx/ISE/14.7/ISE_DS
diff --git a/rtl/xilinx/scarab_ms6p/top/top_xram_sdram.vhd b/rtl/xilinx/scarab_ms6p/top/top_xram_sdram.vhd
index b0ea3d0..e7e0a2c 100644
--- a/rtl/xilinx/scarab_ms6p/top/top_xram_sdram.vhd
+++ b/rtl/xilinx/scarab_ms6p/top/top_xram_sdram.vhd
@@ -41,7 +41,7 @@ entity scarab_xram_sdram is
   generic
   (
     -- ISA: either ARCH_MI32 or ARCH_RV32
\-    C_arch: integer := ARCH_MI32;
\+    C_arch: integer := ARCH_RV32;
     C_debug: boolean := false;
     -- Main clock: 50/81/83/96/100/111/112/125
@@ -53,7 +53,7 @@ entity scarab_xram_sdram is
     C_dcache_size: integer := 8; -- 0, 2, 4, 8, 16, 32 KBytes
     C_cached_addr_bits: integer := 25; -- number of lower RAM address bits 2^25 -> 32MB to be cached
     C_xram_base: std_logic_vector(31 downto 28) := x"8"; -- RAM start address e.g. x"8" -> 0x80000000
\-    C_sdram: boolean := true;
\+    C_sdram: boolean := false;
     -- C_dvid_ddr = false: clk_pixel_shift = 250MHz
     -- C_dvid_ddr = true: clk_pixel_shift = 125MHz

emard, 2016-08-07:
HI

On arduino select also RAM size: 16 KB internal
this sets compiler to make executable which runs from
0x200 origin (there is BRAM) rather than 0x80000000 origin
(the SDRAM)

If that won't work I must see what's broken on RISC-V

roman3017, 2016-08-07:
I have two choices and I have tried both of them with the same result:

32 MB SDRAM onboard
64 KB BRAM internal

There is just a blank screen on both ports and sketches do not work.

emard, 2016-08-07:
HI

OK, it won't work, a proper directory with makefile, separate toplevel
should be made

there's a board tb299 which had spartan-6 lx9 in somewhat different
package where RISC-V was working for me. However, this project
is old and some project files need to be added/removed, it's the best
that I clean up the mess then let's see

however RISC-V is nothing spectacular really, they mis-designed
some parts in ISA so it behaves worse than MIPS

roman3017, 2016-08-07:
It is good to get MIPS working for now. Hope to get RISCV working the next time. Thank you very much for your help, I really appreciate it.

emard, 2016-08-08:
HI

I still can't get RISC-V functional with glue_xram build,
so far, issue 47 "RISC-V broken" is still actual

Same build works if ARCH_MI32 is used instead of
ARCH_RV32

I will assemble bram-only scarab build with all settings
required including text console video for further debugging
until we found out what's wrong there

gornjas, 2016-08-08:
f32c/risc-v currently works only with zero-wait RAM and peripherals. In other words, it won't work with any external RAM configuration, but only with internal BRAMs, hence top_bram and derived designs should have reasonably good chances of booting up as risc-v.

The particular bug which breaks risc-v operation with memory access stalls is hidden somewhere in the branching logic and is difficult to pinpoint and grasp, so we are looking for volunteers willing to find an instruction stream and / or memory access pattern which can be used to reliably reproduce the problem, so that we could try fixing it :)

roman3017, 2016-08-14:
Thank you very much for adding top_xram_bram.vhd. It would not compile as is and I had to add missing files to Makefile and also remove few global arguments from top_xram_bram.vhd. My diff is below. Unfortunately, it still does not work and I am not able to run even the simplest blink sketch as RISCV using 64 KB BRAM internal.

$ git diff
diff --git a/rtl/proj/xilinx/scarab/scarab_ms6p_xram_bram/Makefile b/rtl/proj/xilinx/scarab/scarab_ms6p_xram_bram/Makefile
index 824a908..f5f0230 100644
\--- a/rtl/proj/xilinx/scarab/scarab_ms6p_xram_bram/Makefile
\+++ b/rtl/proj/xilinx/scarab/scarab_ms6p_xram_bram/Makefile
@@ -20,6 +20,9 @@ vfiles = \
 ../../../../soc/pid/simotor_v.v \
 ../../../../soc/pid/rotary_decoder_v.v \
 
+txtfiles = \
+../../../../soc/vgahdmi/vga_textmode_regs.txt \
+
 vhdfiles = \
 ../../../../xilinx/scarab_ms6p/top/top_xram_bram.vhd \
 ../../../../xilinx/scarab_ms6p/clocks/pll_50M_112M5.vhd \
@@ -36,7 +39,6 @@ vhdfiles = \
 ../../../../xilinx/chip/xc6/ddr_out.vhd \
 ../../../../generic/ddr_out.vhd \
 ../../../../generic/glue_xram.vhd \
-../../../../generic/ddr_out.vhd \
 ../../../../generic/bram.vhd \
 ../../../../generic/bootloader/defs_bootblock.vhd \
 ../../../../generic/bootloader/boot_sio_mi32el.vhd \
@@ -47,7 +49,6 @@ vhdfiles = \
 ../../../../generic/bram_true2p_2clk.vhd \
 ../../../../generic/bptrace.vhd \
 ../../../../generic/reg1w2r.vhd \
-../../../../generic/bram_video.vhd \
 ../../../../cpu/alu.vhd \
 ../../../../cpu/loadalign.vhd \
 ../../../../cpu/pipeline.vhd \
@@ -91,6 +92,7 @@ vhdfiles = \
 ../../../../soc/vgahdmi/videofifo.vhd \
 ../../../../soc/vgahdmi/video_cache_i.vhd \
 ../../../../soc/vgahdmi/video_cache_d.vhd \
+../../../../soc/vgahdmi/video_mode_pack.vhd \
 ../../../../soc/vgahdmi/vga.vhd \
 ../../../../soc/vgahdmi/tv.vhd \
 ../../../../soc/cvbs.vhd \
diff --git a/rtl/xilinx/scarab_ms6p/top/top_xram_bram.vhd b/rtl/xilinx/scarab_ms6p/top/top_xram_bram.vhd
index 41a4875..ffd61ee 100644
--- a/rtl/xilinx/scarab_ms6p/top/top_xram_bram.vhd
+++ b/rtl/xilinx/scarab_ms6p/top/top_xram_bram.vhd
@@ -337,10 +337,10 @@ begin
       -- vga simple compositing bitmap only graphics
       C_vgahdmi => C_vgahdmi,
       C_vgahdmi_cache_size => C_vgahdmi_cache_size,
\-      C_vgahdmi_fifo_width => C_vgahdmi_fifo_width,
\-      C_vgahdmi_fifo_height => C_vgahdmi_fifo_height,
\+--      C_vgahdmi_fifo_width => C_vgahdmi_fifo_width,
\+--      C_vgahdmi_fifo_height => C_vgahdmi_fifo_height,
       C_vgahdmi_fifo_data_width => C_vgahdmi_fifo_data_width,
\-      C_vgahdmi_fifo_addr_width => C_vgahdmi_fifo_addr_width,
\+--      C_vgahdmi_fifo_addr_width => C_vgahdmi_fifo_addr_width,
       -- led strip simple compositing bitmap only graphics
       C_ledstrip => C_ledstrip,
       C_ledstrip_fifo_width => C_ledstrip_fifo_width,

roman3017, 2016-08-15:
My statement it does not work may be bit ambiguous, let me be more precise. 

After loading the bit file, I can see the initial fading leds demo on the board working:
xc3sprog -c ftdi miniSpartan6-plus.bit

Afterwards, when I try to load any sketch, I get the following warning and sketch does not work:
.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/WInterrupts.c:309:2: warning: #warning "RISC-V doesn't have interrupts yet. Choose Tools->CPU Architecture->MIPS" [-Wcpp]
 #warning "RISC-V doesn't have interrupts yet. Choose Tools->CPU Architecture->MIPS"
  ^
Sketch uses 976 bytes (1%) of program storage space. Maximum is 64,000 bytes.

roman3017, 2016-08-15:
I apologize, I think my last two comments belong to issue 47 "RISC-V broken".

emard, 2016-08-15:
HI

Yes exactly, RISC-V seems to expose some bug with glue_xram
what happens to you is same to me, bootloader shows fading leds
and it displays prompt, so the bootlader works and it is a tiny RISC-V
code executable, source mostly in C a bit of ASM is in
f32c/src/boot/sio
so you can see what it does

it is actually running, but when upload traffic comes then risc-v somehow
gets buggy

the glue f32c/rtl/generic/glue_bram.vhd should be more suitable for
risc-v instead of glue_xram.vhd,

roman3017, 2016-08-15:
You are right, after replacing glue_xram with glue_bram, sketch blink works as RISCV using 64 KB BRAM internal:
https://github.com/roman3017/f32c/commit/fe356e3a076f8597f2b303a1d34190c14a107b2a
Thank you very much for your help. This should be fine for me while SDRAM is broken.

emard, 2016-08-15:
HI

Ufff I'm glad that at least we have somehow fixed the RISC-V
for you. that is ok for experimentation.

If you discover what is wrong either in glue_xram or glue_bram,
please let us know