f32c/f32c #50: problem with Altera DE0 nano board and f32c(mips)

mrkitravee, 2016-05-30:
I was able to successfully download the bitstream file(de0nano_mips_100mhz.sof)  to DE0 nano board and compile code already , but the result's incorectry . Example we coding LED 5(pin13) output and set to high but the result is LED 1(pin8) and 2(pin9) high . There's except one case that correctly when we don't do anything all of LED was close .

gornjas, 2016-06-04:
I can't reproduce the problem you are describing. Using the standard Arduino "Blink" example, I can select and blink all 8 LEDs, using Arduino logical pin 8 for LED 0, up to pin 15 for LED 7.

If you're concerned with certain LEDs remaining lit after the sketch is downloaded to the FPGA board, that is normal and expected behavior, since blinking the LEDs serves as an indicator of code upload progress. If you'd prefer to turn off all the LEDs when your sketch starts, you should do it explicitly in the setup() function.

emard, 2016-08-21:
Altera DE0 should be working normally now.

closed 2016-08-21 by emard