f32c/f32c #55: Integer multiply bug in interrupt context

emard, 2016-07-29:
Interrupts from arduino mostly unuseable.

Integer "Multiply" instruction should not be executed
from interrupt context, even the simples code has
problems any digitalWrite will use the multply.

dpavlin, 2016-07-30:
Thanks for explanation, I hit this one writing software SPI slave and noticed more-or-less random triggers on pins probably because I was using Serial.println and digitalWrite inside it.

What is safe to use inside ISR like @6bb5aee7e53dc683a0799293bd62836c6d229440

emard, 2016-07-30:
Yes, erratic behaviour is expected. I'll try to convince Marko to fix it.
It's fixable but would consume some LUTs so he though developers
will handle it as-is with c++14 overloading of operator* with expression templates
technique, but arduino user base seems to not like that :). 

First workaround might be to get ISR handler short, and call only those
functions you know won't use multiply. Instead of digitalWrite use
the pointers to registers
led = 13; // arduino pin 13 is led
volatile uint32_t *led_pointer = (volatile uint32_t *) portOutputRegister(digitalPinToPort(led)); // clean way
uint32_t led_bitmask = digitalPinToBitMask(led);

*led_pointer = led_bitmask;

gornjas, 2016-07-30:
I recently tried to add MTHI/MTLO support to the pipeline (required for restoring multiplier state) but given that the multiplier is already our the most timing-critical path, the extra logic further significantly degrades Fmax, even on Lattice XP2 which so far seems to have the fastest 32x32 -> 64 bit multiplier.

I'm on a vacation but give me a few days to try to figure out how to make progress with interrupt-friendly multiplier without sacrificing Fmax (which is already constranied by the multiplier speed on all FPGAs we've touched so far, except the XP2).

emard, 2016-08-12:
Fixed by commits in MTHI/MTLO restore in pipeline.

closed 2016-08-12 by emard