f32c/f32c #56: zrtech board

robotrovsky, 2016-07-29:
Hello!

I have created a Quartus Project for the zrtech Cyclone iv board ( http://www.logifind.com/fpga-cpld-dsp/altera-cyclone-iv-fpga-development-board-ep4ce6e22c8n-1172.html ). The leds are fading in and out, but i am not able to connect to the serial port. Is there anything i could do to determine the reason why it is not working? It would be nice if there would be a debug option, that sends 0x5A or similar continously out the serial port. So i could check if the timing is right. My clock is 48Mhz but i also created with a pll 50 Mhz, setting the the corresponding C_clk_freq to 48 and 50 and passing it as parameter to the glue_bram entity. Because at some points it is written that the frequency should be a multiple of 10Mhz. But neither with 48Mhz nor with 50Mhz SIO is working. This is my config:

	-- ISA
	C_arch: integer := ARCH_MI32;

	-- Main clock freq, in multiples of 10 MHz
	C_clk_freq: integer := 50;

	-- SoC configuration options
	C_bram_size: integer := 24;
	
	C_sio: integer := 1;
	C_sio_fixed_baudrate: boolean := true;
	C_sio_break_detect: boolean := false;
	
	--C_simple_io: boolean := true;

	-- Debugging
	C_debug: boolean := false`

And two more questions: 
1. Is the code for the fading lights in the bootloader? 
2. Is there a way to load a compiled program with the JTAG directly in the BRAM? And if, how do i have to do it in Quartus.

I am a beginner with fpga and would be very thankful for a little support to get my board up and running. Please excuse if this is not the place for questions, but it seems like there is no forum.

gornjas, 2016-07-29:
Fading LEDs confirm that the bootloader is up and running, so you're on the right track!

Re. serial in/out problems, I'd suggest you first try synthesizing a simple design with a loopback between RX and TX coming from / to your PL2303 UART bridge, and confirm that from a serial terminal you can get keystroke responses back from the FPGA. Connecting the serial RX / TX to a LED may help debugging this.

If the loopback test works, perhaps the polarity of SIO signals on PL2303 might need to be inverted, so in your toplevel design you might try inserting a "not" operator between the external RX/TX pins and the glue_bram instance.

Unfortunately I have no idea if and how JTAG could be used to program BRAMs in a f32c SoC.

robotrovsky, 2016-07-29:
Thank you very much! I already did a simple design connecting just RX to TX and the loopback worked. But my error was when i transferred the pinout, because in the official example, from which i imported the pinout, they used it the other way around RX as Output and TX as Input....
I found out, that the pinout was misconfigured by connecting RX to a Led, like you told me. So nice to see the command prompt!! Next time i will double check earlier. This is really a nice project to learn from.

emard, 2016-07-30:
HI

If "program with JTAG the BRAM" you actually want to have your arduino
sketch to get into boot-time f32c executable, then it is possible to
replace the bootloader

compile the sketch with starting address 0 (change in boards.txt 0x200 to 0x0)

Such compiled executable convert with tool
f32c/src/tools/bin2vhdl.sh
or similar hexdump which converts raw binary to comma delimited
vhdl hex like x"1A", x"20", 

with txt editor replace content of "official" bootloader with your stuff
f32c/rtl/generic/bootloader/boot_sio_mi32el.vhd

recompile the bitstream normally and permanent-flash it with
quartus into FPGA config FLASH

Instead of bootloader,  board will run your executable
on power-up.

gornjas, 2016-07-30:
To have a binary reliably executed from a BRAM (ROM) automatically at powerup (instead of the bootloader), we would need to provide a tiny pre-execution routine to set up the stack, and probably also flush the I-cache to be on the safe side.

A dirty kludge could set up a stack as early as possible in main() using __asm __volatile__("lui $29, 0x1000");, but this is not guaranteed to work reliably, as main() on entry might attempt to save registers on an unset stack.

closed 2016-07-30 by gornjas

robotrovsky, 2016-07-31:
Thank you very much.

MehrdadMK, 2021-02-16:
Hi mwyborski

Do you have a verilog code for zrtech to actually read and write onboard Winbond RAM? Thanks

mwyborski, 2021-02-16:
Hi @MehrdadMK ,

i am sorry, i only used the BRAM, you can find it here:
https://github.com/mwyborski/f32c