f32c/f32c #61: Upload works olny 1st time after bitstream

emard, 2016-09-05:
Mostly in vector-enabled bitstreams with external RAM
(sram, sdram, axi) but possibly others might be affected too:

the upload of compiled code works only first time
after the bitstream is uploaded. Subsequent uploads
don't work unless the bitstream is uploaded again.

Workaround is to upload bitstream first before uploading
arduino sketch: (click on arduino->tools->Burn bootloader)

It manifests as if cache is not flushed, but
there might be another reason for this.

emard, 2016-09-29:
Observation:
when interrupts are disabled

C_exceptions: boolean := false;

then upload will not work, problem appears as cache is not working.

When FPGA chip resources are low to fit some projects
then interrupts are disabled 
(vector project on scarab for example).

gornjas, 2016-09-29:
Hmm with ulx2s_sram the problem is observable regardless whether C_exceptions is enabled or not... We seem to be looking for a needle in a haystack...

emard, 2016-09-29:
My workaround for scarab is to reduce from 8 to 4 vectors
and then LUT usage is sufficiently reduced that C_exceptions
can be enabled.
In this case i-cache flush problem disappears.