f32c/f32c #62: Missing vhdl files on Mimas V2 projects

jblang, 2016-09-16:
I am trying to get FPGArduino running on my Mimas V2.

First I tried downloading the prebuilt bitstream but I am getting a 404 when I click the link for Mimas V2 on your website: http://www.nxlab.fer.hr/fpgarduino/numato-mimas-v2.html.

So I decided to check out the code from https://github.com/f32c/f32c and try to build myself.  I opened the project in rtl/proj/xilinx/numato.  I tried mimas_v2_ddr2 both mimas_v2_bram.  When I try to open the project, I get missing files for all the VHDL.  I was able to find almost all of them, but the following are not anywhere in the git repo that I checked out.

s6_numato_mimas_v2_toplevel_bram.vhd
clk_100MHz_25MHz_250MHz.vhd
hdmi_out.vhd
dvid_out.vhd
boot_block_pack.vhd

Any suggestions?

emard, 2016-09-17:
I fixed mimas_v2_bram project to build on my linux:

cd f32c/rtl/proj/xilinx/numato/mimas_v2_bram
make clean; make

I don't have the mimasv2 board, but on positive feedback (you test blink led and some
elementary video output works), I will include the bitstream into mainstream arduino
json installation

Also f32c has axi master interface, so if someone makes DDR2 from
ISE ip core generator and get it to use AXI with f32c, please contribute!

DDR2 project is never actually done, there are driver files
if someone manages to get them compile and running on numato mimas v2
please contribute!

DDR2 driver files are from j2 opensource cpu project which I have seen working 
with DDR2 on numato mimas v2

closed 2016-09-17 by emard

jblang, 2016-09-22:
The Makefile doesn't work on Windows because it has bash commands embedded in it.  I figured out how to get the ISE project to build though by looking at the Makefile and adding the missing files to the project. I submitted pull request #63 for this fix. I tested it with the Arduino IDE and got a few sketches to compile and upload successfully.

Side note, on Windows 10, uploading fails because ftd2xx.dll is missing and ujprog.exe won't run without it. I was able to fix this by downloading the FTDI driver from http://www.ftdichip.com/Drivers/D2XX.htm and putting the DLL in C:\Users\<username>\AppData\Local\Arduino15\packages\FPGArduino\tools\ujprog\1.0.0.  I had to find ujprog.exe and run it manually to see the error dialog about the missing DLL.  The Arduino IDE did not display any errors to indicate that this was the problem so this should probably be documented.

emard, 2016-09-22:
HI

Great, I merged your pull request and left issue for ujprog on windows10
in https://github.com/f32c/tools where ujprog source is

I'm not expert for windows(10 downto 3.11),
but at least for linux the best is to to statically compile
as much as possible in ujprog executable