f32c/f32c #70: HDL simulation

rhinton, 2017-03-17:
I would like to simulate (ModelSim/ISIM/ghdl/etc.) an f32c design.  What is the easiest/best/fastest approach?

rhinton, 2017-03-17:
On a related note, I want to load my program (_main) into BRAM so it just starts up.  Looking at the object dump I get from compiling hello.c, can I just start executing at address x"8000_0000"?

emard, 2017-03-28:
HI

Thanx for your interes and effort!

I never simulated f32c (probably Marko didn't try either :)
Of course RTL core simulation would be cool to have.

The easiest autostart of a simple code might be to modify bootloader
(replace it with your code)
It is 1K or 2K code that makes this serial prompt f32l> which starts from 0 and can
be of any size

Of course mapping something to 0x80000000 could be done, but
bootloader still has to be modified in order to automatically jump there at boot
otherwise command prompt over serial port would be needed

bootloader is here
f32c/src/boot/sio

to setup compiler environment vars of you are on linux/cygwin see this
cd f32c/src/tools
. ./makefiles.sh

compiled binary convert to VHDL with script 
bin2vhdl.sh

and copy-paste generated HEX stuff into bootloader
our default arch is f32c is mips little endian
then this is the bootloader file: 
f32c/rtl/generic/bootloader/boot_sio_mi32el.vhd

This are approximate steps, if you some problem appears
feel free to ask, we'll try to help

closed 2017-08-30 by rhinton

exstob, 2018-04-04:
> compiled binary convert to VHDL with script
bin2vhdl.sh

Can I use compiled binary from Arduino IDE?

emard, 2018-04-04:
Yes, compile arduino sketch for "BRAM" memory target
(Arduino will assign 0x200 start address to the compiled binary)

Then you can replace the bootloader code with the compiled binary
converted to vhdl initialized bram.

easiest way is probably is to fill the first 0x200 bytes (where
original bootloader
is) with NOP's (00) then from 0x200 your compiled binary