f32c/f32c #75: gcc 7 bootloader compilation problem

emard, 2017-11-07:
Oleg Belousov contributed with testing of gcc versions 4 and 7 compilers for the bootloader
and found out

RISC-V 7.1.1 toolchain work well!
Only need to change in src/lib/src/mips/start.S "__gp" to "__global_pointer$" (:

I carefully checked everything and made sure that the 32 bit version of the MIPS 4.8.4 works!
Version gcc 7.x for MIPS does not work, because gcc ignores the attribute((noreturn)). In "_start" prefix:

addiu        sp,sp,-8
sw   ra,4(sp)
sw   s0,0(sp)

For the RISC-V, for some reason it does not matter, but the MIPS sio bootloader does not work properly.
In attach my start.S for 7.1.1 RISC-V toolchain. It's work!

Thank's for your CPU!

/*
 * Clear BSS, set small data index register and jump into main().
 *
 * Assumes that the loader has already properly:
 * 1) set stack pointer
 * 2) set return address
 * 3) invalidated caches
 */
	.section .init
	.globl _start
	.type _start,@function

_start:
	la	gp, __global_pointer$
	la	s1, _end		/* End of BSS section, word aligned */
	la	s0, __bss_start		/* Start of BSS section, NOT word aligned */

	la	tp, _end + 63
	and	tp, tp, -64

bss_clear:
	sw	zero, 0(s0)
	addi	s0, s0, 4
	blt	s0, s1, bss_clear

	la	s0, __init_array_start
	la	s1, __init_array_end
	move	s2, ra
	j	ctor_loop_enter

ctor_loop:
	lw	a0, (s0)
	addi	s0, s0, 4
	jalr	a0
ctor_loop_enter:
	bne	s0, s1,	ctor_loop
	move	ra, s2
	j	main

strijar, 2017-11-08:
There I have a typo. To, of course, work with riscv, you need to fix its file - src/lib/src/riscv/start.S

emard, 2017-12-06:
Should be fixed in our latest gcc 7.2 that automatically
gets installed with fpgarduino

closed 2017-12-06 by emard