f32c/f32c #77: src/examples won't compile

emard, 2017-12-21:


emard, 2017-12-22:
this is because f32c libraries are no longer distributed in binary form
and should be compiled from source

exstob, 2018-07-13:
I try to compile example in folder "sdcard" and get next error:

user@server /c/MinGW/msys/1.0/home/uskov.o/f32c/src/examples/sdcard
$ make
riscv32-elf-ld -N -EL --section-start=.init=0x00000400 --library-path=/c/MinGW/msys/1.0/home/uskov.o/f32c/src/lib/riscv -gc-sections  -o sdtest ./obj/riscv/../../lib/src/riscv/start.O ./obj/riscv/sdcard_test.o ./obj/riscv/../../lib/src/sio_poll.o ./obj/riscv/../../lib/src/printf.o ./obj/riscv/../../lib/src/fatfs.o ./obj/riscv/../../lib/src/file.o ./obj/riscv/../../lib/src/malloc.o ./obj/riscv/../../lib/src/spi.o ./obj/riscv/../../lib/src/sdcard.o ./obj/riscv/../../lib/src/diskio.o ./obj/riscv/../../lib/src/mul.o ./obj/riscv/../../lib/src/div.o
./obj/riscv/../../lib/src/printf.o: In function `.L82':
printf.c:(.text._xvprintf+0x4b0): undefined reference to `__umoddi3'
printf.c:(.text._xvprintf+0x4dc): undefined reference to `__udivdi3'
make: *** [sdtest] Error 1

As I understand, this error for a reason using type uint64_t in file "div64.c", but how fix it?

exstob, 2018-07-30:
I solved this problem. Need simply add path to "div64.c " in Makefile.

goran-mahovlic, 2023-04-29:
Solution for:

```
mips-elf-ld: cannot find -lcrt0
mips-elf-ld: cannot find -lcint
```

https://github.com/f32c/f32c/issues/68#issuecomment-287402183

```
$ cd <path-to>/f32c/src/lib/src
$ make all_libs
```

So we can close it...

closed 2023-04-29 by goran-mahovlic