f32c/f32c #79: f32c for ECP5 crashes latest diamonds

emard, 2017-12-22:
Diamond 3.7 correctly compiles ECP5 f32c bitstreams so I haven't upgraded.
I would fix it f32c source but can't locate what is causing it.

I think diamond versions 3.8 to 3.10 after few minutes of compiling crash with
some internal segmentation fault asking to contact synopsys, without any
information where in the source the problem appears, like vhdl file name and
source line number.

MayaPosch, 2019-07-19:
It appears that with Diamond 3.11 Synplify Pro no longer crashes.

emard, 2019-07-19:
I still use diamond 3.7 because it successfully compiles f32c to
operate at 100 MHz. 3.11 even when it doesn't crash in my testing
was able to compile f32c only for up to 83 MHz, and couldn't reach 100.

Inferring of signed multiply (used in 128-voice synth) has same bug as 3.7
and I have done workaround in vhdl source but it's
one more reason to be slightly skepical about replacing 3.7 with 3.11

emard, 2019-10-31:
Latest diamond 3.11 seems to work and compile f32c
normally, it is useable but I still decline to switch 3.7 to 3.11
because I haven't yet seen any example where 3.11 does
better binary than 3.7

closed 2019-10-31 by emard