f32c/f32c #83: Expression error on DE10-lite

petit-miner, 2019-01-26:
Hi there,

I'm unable to build the core for the MAX10 on the DE10-lite. First of all the project file misses a lot of needed libraries. I added them manually, but now I get this error:

Error (10344): VHDL expression error at top_de10lite_xram_sdram_vector.vhd(189): expression has 16 elements, but must have 32 elements
Error (10344): VHDL expression error at top_de10lite_xram_sdram_vector.vhd(191): expression has 2 elements, but must have 4 elements
Error (10344): VHDL expression error at top_de10lite_xram_sdram_vector.vhd(191): expression has 4 elements, but must have 2 elements

Can somebody help me with this issue?

va1ery, 2021-12-15:
Same Error (10344): VHDL expression error at top_de10lite_xram_sdram_vector.vhd(191): expression has 4 elements, but must have 2 elements
Have sent email to authors and still there is no reply

va1ery, 2023-04-16:
Fixed here https://github.com/va1ery/f32Arduino[Link](https://github.com/va1ery/f32Arduino)