f32c/f32c #91: Multicore in FPGArduino

carlosdelfino, 2020-07-06:
I am developing studies with [FreeRTOS](https://freertos.com) and applying it in [DuinOS](https://github.com/DuinOS), and I would like to expand such studies with a Multicore environment, I initially thought about investing in the [Arduino Potenza H7](https://blog.arduino.cc/2020/01/07/arduino-goes-pro-at-ces-2020/) that has a Cortex-M7 core and a Cortex-M4 core, it would be very interesting to develop solutions using Message Queue between Threads.

But when I got to know the FPGArduino, I see that it would be much more interesting to apply such studies adding to the Risc-V and FPGA.

Is there any possibility of expanding the project to multi-core? two or more?

thankful

gornjas, 2020-07-06:
The f32c core which is in the focus of the fpgarduino project currently lacks atomic memory primitives (LL/SC) and cache coherency circuitry. The first step would be adding those bits, which is not entirely trivial.

carlosdelfino, 2020-07-06:
@gornjas , I am a beginner in the topic.

In the case in addition to the core fpga, would changes have to be made to the GCC-f32c?